polpo/picogus #1: Tested a few more demos

zonaj, 2022-12-04:
I've tested a few unverified demos.

Seem perfect:
- Stash by The Black Lotus
- Fluid Motion by Valhalla
- Heartquake by Iguana
- DX Project by RealTech
- My Summer Demo by Byterapers
- Solstice by Valhalla
- Caero by Plant and EMF
- Aquaphobia by RealTech
- Contrast by Oxygene
- Machines of Madness by Cubic Team
- Lasse Reinbøng by Cubic Team & $eeN
- Genocyd by GMF
- Mindtrap by Trauma
- Jizz by The Black Lotus
- Robotnik by Rage
- Boost by Doomsday

Imperfect:
- Disgust by Camorra (credits at the end have strange timing)

Possibly problematic (I did not try very hard to run them):
- Sunflower by Pulse (hangs at the beginning)
- Pump by Iguana (hangs in the middle)

polpo, 2022-12-11:
Thank you for all of the results! I will add them to the compatibility list.

polpo, 2022-12-11:
I'm happy to report that Sunflower by Pulse runs perfectly now with my latest firmware that I will be releasing soon 😀

polpo, 2022-12-12:
The compatibility list has been updated with these demos. Thanks again for opening this issue!

closed 2022-12-12 by polpo

zonaj, 2022-12-13:
Very cool, I confirm that sunflower works with the latest firmware! Same thing with "Pump" by iguana. Works great now.