polpo/picogus #3: Potential bracket options

tyrells, 2023-02-03:
I was just doing some searching for other potential back bracket options:

- https://www.mouser.com/ProductDetail/Keystone-Electronics/9203-1?qs=%252BU2dxLPlMC27%252BsH2ZGD%252BtA%3D%3D
- https://www.aliexpress.com/item/1005001466248688.html

The screw holes are in different places, so this would need a PCB change, but that may be a good option assuming the Keystone Electronics 9202-1 bracket isn't coming back in stock.

polpo, 2023-02-05:
Unfortunately those brackets are made for PCI cards so the mounting holes are on the other side of the bracket. Standoffs might solve that problem, though...

tyrells, 2023-02-06:
I didn't realize that. Thank you. I will keep looking.

closed 2023-02-06 by tyrells

arkayii, 2023-02-07:
If there are people who have zero other viable bracket options, I would be happy to accurately drill out a 9202 bracket and send it out, but the necessary cost of international shipping means it would only be a viable option for those who are particularly keen - I'm guessing the total cost would be at least $10, if I did it free, which I would. Nonetheless feel free to contact me here, or on Vogons as Shreddoc.