polpo/picogus #4: 2508051027Y1 (L1/L2) substitute?

madcatse, 2023-02-03:
I have serious trouble finding 2508051027Y1 ferrite beads where I live, and I have no option of ordering from Mouser. Can I use BLM21SP102SN1D instead? It has maximum DC Resistance of 120 mOhms instead of 300 mOhms. Will it do?

ramapcsx2, 2023-02-05:
Hey,
I'm not at all part of this project, but it seems to me that these ferrites are just filtering for an (optional) MIDI or UART port.
You should be fine placing your weaker ferrites there, no problem :)

polpo, 2023-02-05:
+1, that value should be fine. The MIDI electrical specification says the ferrites are optional, so in a pinch you can bridge L1 and L2.

madcatse, 2023-02-05:
Thank you! I've managed to source the required ferrites in the end, but they will take 10-12 weeks to arrive in any case. >_<