polpo/picogus #7: I have no knowledge of electronic circuits, but I would like to make your project.

drawingangnam, 2023-03-11:
Firstly, thanks for sharing your awesome project.

I have no knowledge of electronic circuits, but I enjoy soldering.

I would like to give your project as a gift for my homemade retro PC, but some IC chips are hard to find in my country.

74CB3T3257PWR
74CB3T3245PWR

Is there a replacement for the above two chips?

If there are compatible chips, please advise.

Thank you.

drawingangnam, 2023-03-12:
I will find oversea marcket.

closed 2023-03-12 by drawingangnam

ramapcsx2, 2023-03-12:
These seem a bit special (I haven't heard of them before), though they carry the '74' identifier. '74' chips are supposed to be widely available off the shelves chips.
Best of luck!

polpo, 2023-03-13:
The 74CB3T3245PWR can be replaced with a TSSOP-14 74LVC244. I have tested the Nexperia 74LVC244APW,118 and it works fine.

The 74CB3T3257PWR is harder to find a replacement for. I have not found another 2:1 mux/demux chip that also does level shifting. The Pico unofficially can tolerate 5V signals, so another 3257-type mux could work, but I have not tested any alternatives. Options could include the Nexperia 74CB3Q3257PWJ but that would pass 5V through to the Pico. The Analog Devices ADG3257 is also interesting, but without bodges on the board it would pass 2.6V in both directions which is kind of borderline for TTL logic levels.

ramapcsx2, 2023-03-13:
(Any kind of sources for that 5V tolerance? I realize they wouldn't print it in the spec sheets, but maybe a hint on a forum post or so :p)

polpo, 2023-03-13:
@ramapcsx2 sorry, I forgot to link it. Eben Upton himself mentions it [in this Hackaday comment](https://hackaday.com/2022/09/25/pi-pico-w-does-pcmcia-gets-this-ibm-pc110-online/#comment-6516402):

> 
> Note: the pads themselves are 5V tolerant for input, just not 5V5 tolerant, which is what we’d need them to be to put “5V tolerant” in the data sheet. (Hypothetical) future designs will fix this.
> 
> Note also: they’re not (and will never be) 5V *failsafe*, so don’t subject them to 5V without first powering VDDIO.

ramapcsx2, 2023-03-13:
Epic, thank you! :)