polpo/picogus #8: The FX sound is not coming out.

drawingangnam, 2023-05-13:
Hello! 
I have finally completed your wonderful project by myself. 
Although there was a minor issue during production, I was able to resolve it successfully. 
Instead of using the 74CB3T3245PWR, I used the 74LVC244, and the card booted up well.

The card is recognized well in pgusinit and sounds great in adlib mode. 
However, when I run Duke Nukem or Doom in GUS mode, the music plays well, but the FX sound doesn't come out.

Do you have any advice?
 I've tried various jumper settings and experimented with different computers, but the symptoms are the same.
 I also tried different settings such as "set ultrast=240,3,3,7" after using "set ultrast=240,1,1,5,5" (and, of course, I set the jumper on the board accordingly)."

drawingangnam, 2023-05-13:
Oh... Finally, the FX sound is working on the third 586 PC. 
I don't know why there was no sound in the FX or MIDIDEMO.BAT on the first and second computers. 
I'll try to solve the problem slowly.

Also, there is no sound in Sound Blaster emulation mode. 
I don't know what I did wrong. 
When I run E.BAT and choose "1 = Sound Blaster emulation", 
it shows "GF1 SBOS V3.82 The NMI procedure on this P.C. is disabled.

When I run MIDIDEMO.BAT, the MIDI sound works well at first, but then it stops in the middle. This happens with all other songs as well. (It stops when one note is sustained for too long.)

Could you please give me some advice?

polpo, 2023-05-13:
Sound FX are streamed in via DMA into a pair of looping channels, and it requires really tight timing between IRQ and DMA signals to work well. I've tuned the DMA support on motherboards with Intel 430FX and 430VX chipsets, and support for other chipsets is hit-or-miss because I only have a few motherboards to test on. See [the general compatibility notes](https://github.com/polpo/picogus/wiki/Compatibility-list#general-compatibility-notes) in the wiki to see what is reported as working and not working. What chipsets did you not get sound FX on, and what chipset does the third 586 PC have?

Unfortunately, SBOS does not work and will never work on the current hardware revision. The pin that generates an NMI (IOCHCHK) is not wired on the PicoGUS PCB and SBOS requires it to operate. I should add a note to the compatibility list wiki page about that.

Support of Gravis's PlayMIDI (which MIDIDEMO.BAT runs) is not very good, unfortunately. What you report is actually a bit better than what I've seen on my own motherboards, which also include corrupted sounding samples. I haven't been able to make much progress in improving PlayMIDI. If you want to play MIDI, try DOSMID with ULTRAMID.EXE loaded. It should work much better.

drawingangnam, 2023-05-14:
> 
Your prompt and informative response is greatly appreciated as well.

As a hobby, I am currently building a retro gaming PC using a 386-586 compatible industrial PC in the PC104 form factor.
Although I do not possess any professional knowledge in electrical engineering, I will do my utmost to share compatibility information between the PC104 computer and PICOGUS.

As you mentioned, I ran Ultramidi and attempted to play Dosmid, but I received an error message indicating that there was no MIDI device detected. I was curious if ULTRAMID or DOSMID requires any specific options or settings?
Could you provide any insight into this issue?

Thank you once again for your remarkable efforts and contributions to the world of retro PCs.

ThatRed, 2023-05-21:
Hi drawingangnam, 

I'm currently building a Dolch PAC inspired retro PC, based on a PICMG SBC board. It's a Pentium I 200 MHz with PC104 female headers. If I could just plug the PicoGUS directly into the PC104 headers, I could save an ISA slot... 

Did you just solder pin headers so they stick out of the back of the PicoGUS pcb and plugged it into your PC104 board, that the top side with the Pi Pico is visible or did you have to make some adjustments?

Thanks in advance and sorry for hijacking this thread!

closed 2023-10-05 by drawingangnam