polpo/picogus #9: When flashing from PG-GUS.UF2 0.6.1, I get: ERROR: Card is not in firmware writing mode?

ThatRed, 2023-05-21:
Hi, thank you for this very cool project!

When I try to flash from PG-GUS.UF2 0.6.1 to any different firmware in DOS, I get "Programming........ ERROR: Card is not in firmware writing mode?" and after that the card is "bricked". If I run pgusinit again, I get: "ERROR: no PicoGUS detected!"

If I copy a firmware from USB that is not PG-GUS 0.6.1, flashing from firmware to firmware works from DOS without any problems, unless it is PG-GUS.UF2 0.6.1...

Can you please take a look? Thanks.

polpo, 2023-09-29:
Hi, sorry for taking so long to reply. I'm interested to know if you still have this problem with the 0.7.0 firmware.

If you still have this problem, what CPU/motherboard/chipset do you have? Someone else has reported this same issue to me with the 0.7.0 firmware and I'm interested to know if you have a chipset in common.

drlarsen77, 2024-07-26:
I'm not sure if this helps anyone, but I ran into this issue/symptoms upgrading from 1.2.0 to 2.0.0. It turned out the AT clock speed was inadvertently set wrong. After setting it to 8 MHz, it started working normally.