polpo/picogus #11: New good chipsets tested

zxrepo, 2023-07-27:
Hi, Ian!
Thank you so much for your amazing project! Just assembled and tested PicoGUS. I would like to share my results:

1) KM-S4-1 motherboard (aka U6923, Rectron RT-4s3) SiS 85C496/497 chipset, am486DX4-100
https://www.elhvb.com/webhq/models/486pci/km-s4-1.htm

2) Asus CUV4X motherboard based on VIA Apollo Pro133A chipset, PIII-733
http://hw-museum.cz/mb/46/asus-cuv4x
excellent results but we have to setup a couple of settings in BIOS before:
'Advanced'->'PCI/PNP ISA IRQ Resource Exclusion'->'IRQ 5 Used by ISA' set to 'Yes'
'Advanced'->'PCI/PNP ISA DMA Resource Exclusion'->'DMA 1 Used by ISA' set to 'Yes'

Hope you will find some solution for SBOS and NMI issue someday somehow for all of us! )) Thanks again!!! Cheers!

polpo, 2023-07-31:
@zxrepo thank you very much for these reports! Seeing that the VIA Apollo Pro 133A chipset works is good news, as their MVP3 chipset has issues with DMA and I was worried that it carried onto other, later chipsets. Have you tested if it works in Doom (versions greater than 1.2)? That's an example when the PicoGUS falls over with the VIA MVP3 chipset and the game fails to start.

zxrepo, 2023-08-01:
@polpo  I tested on Ultimate Doom (i.e. version 1.9) - https://youtu.be/VKvyhqkmX5s?t=410
It worked like a charm. By the way, Doom works fine even without proper BIOS PCI/PNP ISA IRQ/DMA settings (unlike other games such as One Must Fall: 2097 e.g.). In other cases it's mandatory to setup IRQ 5 and DMA 1 Used by ISA.

polpo, 2023-08-01:
Wonderful! Thanks for the video, it sounds great! I'll update the compatibility list with these chipsets.

polpo, 2023-09-27:
Compatibility list updated! Thanks again.

polpo, 2023-09-30:
@zxrepo I have a report that updating firmware from DOS with pgusinit on a 486 motherboard with the SiS 85C496/497 chipset fails. Do you know if it works for you?

zxrepo, 2023-09-30:
@polpo just tested this case. And I got exactly the same results for my SiS85C497 motherboard unfortunately.
Firmware update does not occur. The message is 'Card is not if firmware writing mode?'
error message: https://i.imgur.com/AHqqfGf.jpg
installed audio card: https://i.imgur.com/z1I1XKW.jpg

For the same PicoGUS board and my another motherboard asus-cuv4x it's all ok. The firmware update process is successful for VIA694X/694Z AGPset/VT82C686A set.

polpo, 2023-09-30:
Ok, thank you very much for testing this! I've updated the compatibility list with this detail.

closed 2023-09-30 by polpo