polpo/picogus #13: Chipset Compatibility

Xenenic, 2023-09-05:
Tested with the same image on each motherboard with the same PicoGUS setting.
[OS] 
- DOS 6.22

[PicoGUS] 
- FW v0.6.1 
- IRQ 5 
- DMA 1
- PCB v1.1.1

[Test Games]
DOOM/DUKE3D/DECENT/HERETIC/COMMAND AND CONQUER

[Socket | Chipset | Motherboard | Notes]
[Perfect]
- Socket 3 - UMC UM8498/8496 - PCChips M912 v1.7 - https://theretroweb.com/motherboards/s/pcchips-m912-v1.7-486
- Socket 3 - UMC UM8498/8496 - Biostar MB-14** ??? (UMC498F 2VLB) - https://theretroweb.com/motherboards/s/biostar-mb-14-umc498f-2vlb
- Socket 5 - Intel 430TX (PCIset TX Triton II) - ECS P5TX-Bpro - https://theretroweb.com/motherboards/s/ecs-p5tx-bpro

[Works With Issues]
- Super Socket 7 - SiS 530/5595 (Sinbad) - Soyo SY-5SSM V1.1 - Random hangs/crashes https://theretroweb.com/motherboards/s/soyo-sy-5ssm-v1.1
- Super Socket 7 - VIA VT82C598MVP (Apollo MVP3)- DFI K6BV3+ (REV.A+) - Sounds are randomly 'tincan like' or have static https://theretroweb.com/motherboards/s/dfi-k6bv3-rev.a

[Does Not Work]
- Socket 5 - PCISet SB82371FB - UNKNOWN - PGUSINIT causes hang

polpo, 2023-09-11:
Thank you so much for these detailed reports! I'll add them to the compatibility list. I now have a few more systems to add to my test bench. I've recently acquired a board with a VIA Apollo MVP3 chipset and also had problems. However this system has the same problems with my real GUS which is interesting.

The system with the PCISet SB82371FB - that's the PIIX south bridge for the 430FX chipset. That's the primary chipset that I develop PicoGUS on so I wonder if something is odd with your motherboard. Is the ISA bus speed configurable in the BIOS for this motherboard? Any other bus timing configuration?

Xenenic, 2023-09-12:
You're welcome! I'll get a scope on the PCISet SB82371FB to verify the ISA bus is at/below 8.33MHz but per the BIOS settings it should have been. I'll also add a step to measure the ISA bus going forward and for the boards I've tested above which had issues or failed.

Xenenic, 2023-09-12:
Following up on the PCISet SB82371FB the issue is related to the board itself, ISA is clocking at 11~MHz regardless of BIOS settings.
[image]

polpo, 2023-09-15:
Oh yikes! Yeah, the PicoGUS has no chance of keeping up with an 11MHz ISA bus. That's too bad that BIOS settings have no effect on the speed. Fortunately my system with that same southbridge has a fixed 8MHz ISA bus speed. Hopefully you can figure out the make/model of that particular motherboard.

Xenenic, 2023-09-16:
> Oh yikes! Yeah, the PicoGUS has no chance of keeping up with an 11MHz ISA bus. That's too bad that BIOS settings have no effect on the speed. Fortunately my system with that same southbridge has a fixed 8MHz ISA bus speed. Hopefully you can figure out the make/model of that particular motherboard so I can call it out specifically on the compatibility list.

Tracked down the board to the generic "OEM Gateway 2000 P5" series, my guess would be the P5-66/100 with the board supporting up to a Pentium 133MHz via undocumented settings.

polpo, 2023-09-27:
Updated the compatibility list with these reports. Thanks again!

closed 2023-09-27 by polpo