polpo/picogus #15: Cannot compile project

maxgerhardt, 2023-10-08:
On Windows or Linux it's the same thing: When I download 
* Pico SDK latest stable 1.5.1
* Pico-Extras matching branch 1.5.1
* Arm-none-eabi-gcc 10.3.1 (10.3-2021.10)
* CMake  3.25.1
* Ninja 1.11.1

And compile the project, it fails to find a bunch of implementation for linker-wrapped functions at the end, specifically STDIO and Math ones 
* `__wrap_puts`
* `__wrap_printf`
* `__wrap_remainder`

```
[ 96%] Building ASM object CMakeFiles/picogus.dir/home/runner/work/picogus/picogus/pico/pico-extras/src/common/pico_audio/audio_utils.S.obj
[ 97%] Building C object CMakeFiles/picogus.dir/home/runner/work/picogus/picogus/pico/pico-extras/src/common/pico_util_buffer/buffer.c.obj
[ 98%] Building C object CMakeFiles/picogus.dir/home/runner/work/picogus/picogus/pico/pico-sdk/src/rp2_common/hardware_flash/flash.c.obj
[100%] Linking CXX executable picogus.elf
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /tmp/picogus.elf.RtR3fl.ltrans0.ltrans.o: in function `_out_char':
<artificial>:(.text+0x1f90): undefined reference to `__wrap_putchar'
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /tmp/picogus.elf.RtR3fl.ltrans0.ltrans.o: in function `panic':
<artificial>:(.text+0x2150): undefined reference to `__wrap_puts'
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: <artificial>:(.text+0x2158): undefined reference to `__wrap_puts'
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: <artificial>:(.text+0x215e): undefined reference to `__wrap_puts'
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /tmp/picogus.elf.RtR3fl.ltrans0.ltrans.o: in function `play_gus()':
<artificial>:(.text+0x2b40): undefined reference to `__wrap_puts'
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: <artificial>:(.text+0x2bfc): undefined reference to `__wrap_puts'
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /tmp/picogus.elf.RtR3fl.ltrans0.ltrans.o:<artificial>:(.text+0x2f84): more undefined references to `__wrap_puts' follow
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /tmp/picogus.elf.RtR3fl.ltrans0.ltrans.o: in function `play_gus()':
<artificial>:(.text+0x2ff8): undefined reference to `__wrap_printf'
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: <artificial>:(.text+0x39aa): undefined reference to `__wrap_printf'
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /tmp/picogus.elf.RtR3fl.ltrans0.ltrans.o: in function `__wrap_remainder':
<artificial>:(.text+0x3bb8): undefined reference to `__wrap_remquo'
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /tmp/picogus.elf.RtR3fl.ltrans0.ltrans.o: in function `__wrap_remainderf':
<artificial>:(.text+0x3c34): undefined reference to `__wrap_remquof'
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /tmp/picogus.elf.RtR3fl.ltrans0.ltrans.o: in function `main':
<artificial>:(.text.startup+0x19a): undefined reference to `__wrap_puts'
/home/runner/gcc-arm-none-eabi-10.2020.4-linux/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: <artificial>:(.text.startup+0x200): undefined reference to `__wrap_puts'
```

Which version of the SDK and compiler etc. are you using to build this project? The Wiki and README does not mention it.

maxgerhardt, 2023-10-08:
I'm also quite a bit confused on how files like `pico_reflash.c` (which is compiled in every case) can do `printf()` when e.g. in the OPL project enviornment, all STDIO functions are turned off.
```cmake
    pico_enable_stdio_uart(picogus 0)
    pico_enable_stdio_usb(picogus 0)
    pico_enable_stdio_semihosting(picogus 0)
```

polpo, 2023-10-08:
Sorry, better documenting building is something I need to do. I use LTO which does not work out of the box with the Pico SDK, so I've patched the SDK to allow for it. If you want to compile without needing to patch the SDK, disable LTO by removing [this line from CMakeLists.txt](https://github.com/polpo/picogus/blob/56ad321f988d3ec19579297da167bd6c8202a079/sw/CMakeLists.txt#L34C1-L34C37):

```
add_compile_options(-flto=jobserver)
```

I use the latest versions of the ARM GNU toolchain and Pico SDK.

Disabling stdio in that way doesn't remove stdio completely, it just makes stdout/err/in go nowhere.

I forgot that I turned all stdio off in OPL... it's been a while since I touched that part of the project. I don't remember why I did that! I do purposefully turn it off for MPU, so maybe it's just a copy-and-paste from there.

maxgerhardt, 2023-10-08:
I've just confirmed that this works indeed and the .uf2 file is now built.

If LTO is necessary for the project (size or performance wise), can you push your modified Pico-SDK? It does not seem to be in https://github.com/polpo?tab=repositories&q=sdk&type=&language=&sort=.

>I use the latest versions of the ARM GNU toolchain 

Just to confirm, that is arm-none-eabi-gcc version 12?

polpo, 2023-10-08:
I don't consider LTO really necessary. I need to do some real comparisons to see if it's of measurable benefit. I manually set most functions as inline almost everything anyway for performance reasons.

If you want to see what LTO does, change [this line in platform.h in the Pico SDK](https://github.com/raspberrypi/pico-sdk/blob/6a7db34ff63345a7badec79ebea3aaef1712f374/src/rp2_common/pico_platform/include/pico/platform.h#L487C1-L487C37) from:

```
#define WRAPPER_FUNC(x) __wrap_ ## x
```

to

```
#define WRAPPER_FUNC(x) __attribute__((used)) __wrap_ ## x
```

> Just to confirm, that is arm-none-eabi-gcc version 12?

Correct. I am using arm-none-eabi-gcc-12.3.1 currently but any version 12 will work fine.

polpo, 2023-11-07:
Fixed in your GitHub action MR and improved in c475db6 - LTO is no longer enabled by default and can be enabled by adding a `-DUSE_LTO` on the CMake command line.

closed 2023-11-07 by polpo