polpo/picogus #18: Digital s/pdif out options

skadarnold, 2023-10-09:
I am loving this project at the moment, and plan to build a board and do some of my own testing. Awesome work! I am wondering if there were any thoughts or plans to implement a digital output option? It seems it would be pretty easy to replace the PCM5102 with an I2C to spdif converter. But I am wondering if there were any thoughts for either building a dedicated option for spdif coax, or at least a pinout to allow connection to a PCI breakout board. The PCM5102 does a great job in my experience, but I try to use spdif out as much as possible to help reduce ground noise generated from the computer internally.

polpo, 2023-10-09:
The pico-extras library I use for I2S audio out also has a SPDIF audio out module, so integrating that would actually be pretty simple! And the UART TX header on the PicoGUS board could be used for TTL SPDIF instead, and converting that to TOSLINK or coax on a breakout board like you suggest would be pretty simple. The hangup would be around non-standard sampling rates which I'm assuming SPDIF gear would not be happy with. The GUS can change sampling rate among 18 different values between 44100Hz and 19293Hz almost constantly, and the OPL2 emulation also uses a strange sampling rate of 49716Hz. There might be a way to force to 44100Hz for GUS, but I haven't looked too much into it as it may be too much for the Pico or PSRAM to handle.

Anyway I'll keep this issue open if I get fixed 44.1kHz sampling rate to work.

skadarnold, 2023-10-09:
This is great news, especially considering the current UART TX on the board could already allow for this expansion. Having a dedicated hardware module for this would be a nice option as well for the future, but using the UART we would have the option to do some testing on this current hardware revision. 

That is very interesting about the non standard sample rates as well. I would be curious to see how external DAC's would handle  the changes.  Does the PCM5102 over I2S behave different than TTL over SPDIF in reference to these odd sample rates?