polpo/picogus #20: SCK Short to Ground - I2S DAC Questions

skadarnold, 2023-11-03:
I am having issues with my PicoGUS detection whenever I insert the 3.5mm line into the I2S DAC module. This is consistent so I have been looking for problems in that area. The SCK line seems to be shorting to ground. All of the joints look clean, but it makes me wonder if I should have left out soldering certain pins that were undefined in the build guide. I will continue to troubleshoot and will probably pull the I2S DAC out to get a better look at things. 

I question whether it was incorrect of me to solder in these extra pads along the length of the module? I had an extra pin header to that length and thought it wouldn't hurt to solder them down for extra support, but maybe I should have only soldered LGRG or finished soldering the whole row down? 

[image]

polpo, 2023-11-03:
Hi, all of those extra pins are not connected anywhere on the PicoGUS board so soldering them should not cause any issue, unless you've somehow caused a short with a solder bridge or something... And SCK is intentionally grounded because the PicoGUS uses the PCM510xA chip in system clock PLL mode, so that's expected.

After hearing from people's experiences with the purple DAC modules on the PicoGUS boards they've built, I've learned that they have pretty spotty QA. There's a chance that your module is bad.

skadarnold, 2023-11-03:
Thanks for the fast reply. This is all good information to know. I was able to remove the DAC module so I will continue to test. I might even socket the DAC module this time so I can test modules easier. I will keep in touch to what I discover.

polpo, 2023-11-03:
No, LGRG are not required. They're only there for physical support.

skadarnold, 2023-11-03:
I had a moment to strap a new module on it. At first it seemed to have fixed it, but then further testing shows that it is functioning about the same. The first time I tried it with 3.55mm audio inserted it worked just fine, but was inconsistent afterwards. With that line inserted there is about 1/6 or less chance of the PicoGus initializing. I also noticed that with the line out, the light on the pi pico stays on and steady when booting the system. As soon as the 3.55mm is inserted the light will only stay on for a few moments at boot. This might be a general compatibility thing with this motherboard and PSU combo, I will have to do some further testing to know for certain. I will also try different 3.55mm cables to see if there is something specific.

skadarnold, 2023-11-03:
So I am finding that the DAC module and inserting the audio cable is not as relevant as I first thought. It maybe has a slight affect, but even without it the pi pico will only light up about half of the time when I start the computer. I have tested this on two different machines which seems to have identical behavior on both of them.  Both motherboards tested are different variations of AG430HX but I will try and confirm whether or not it is a motherboard compatibility and issue with my PicoGus

arkayii, 2023-11-03:
Any chance it's related to the reset line instability/sensitivity that has been seen in other Picogus contexts recently? - probably not, but didn't want it to go unmentioned.

Also, when doing my own Picogus builds, I too was uncertain at times, about the expected status of the LED light on the Pico. All I knew was that an unstable build was often accompanied by inconsistent Pico LED lighting, and that thorough (solder) checks and reflows fixed almost all of the issues of this nature that I encountered. That said I certainly wouldn't discount other possible causes, such as that I initially mentioned.

skadarnold, 2023-11-03:
> Any chance it's related to the reset line instability/sensitivity that has been seen in other Picogus contexts recently? - probably not, but didn't want it to go unmentioned.
>
> Also, when doing my [...]

Good question about the reset line. I haven't seen any other posts about it, but it also crossed my mind that the pico was stuck in reset. I am looking over all my solder work as I have also been wondering if something isn't quite connecting correctly.

arkayii, 2023-11-03:
> Good question about the reset line. I haven't seen any other posts about it, but it also crossed my mind that the pico was stuck in reset. I am looking over all my solder work as I have also been [...]

The soldering at the 3 x TSSOP chips are common culprits, for tiny bridging issues. Yours may be fine, but insofar as anywhere is worth checking, that is certainly one area.

Regarding the reset line fixes, there are some forum posts on Vogons (starting around here : https://www.vogons.org/viewtopic.php?p=1205085#p1205085 ) where a capacitor fix has been used by some people, e.g. in the context of Book8088/Hand386 devices. Whether or not this is relevant to your or any other situation, I do not know, but let it not be said that it's gone unmentioned! :)

skadarnold, 2023-11-04:
Well I think I have exhausted all of my troubleshooting for today. I went through and reflowed everything 2 - 3 times now, everything looks great and I am pretty certain there are no micro bridges at this point. I also added a 100nf capacitor as suggested in the vogons thread to help fix any reset bugs as mentioned. Nothing seems to change its behavior. Sometimes it will give me a working streak and initialize perfectly for about 3 reboots, then it stops for a while and will randomly work again for no real reason. 

I can see that when it does work, the light on the pico will flash on briefly, maybe a couple of times and then remain solid on after initializing. But if I do not see those first couple of flashes when the computer boots then it will not initialize. 

I will give it a try in my 440bx motherboard and see if it behaves differently.

polpo, 2023-11-04:
If you want to completely eliminate the variable of a noisy/glitchy reset line on your motherboard, you can cover the reset pin (B2) on the ISA connector with an insulating tape (Kapton would be best), then use the reset jumper that I see you have to ground the reset line on the PicoGUS board. The pin in question is circled in yellow in this photo:

[image]

Hopefully the changes I'm introducing in my new designs like switching to a 74AHC14 chip and adding an RC on the RUN pin will eliminate these sorts of issues. If doing the above solves your issue, you can cut the trace going to the reset pin on the ISA connector. The only real downside is that any notes playing will remain stuck playing when you hit reset on your computer, until you run pgusinit to reset the card.

skadarnold, 2023-11-04:
> If you want to completely eliminate the variable of a noisy/glitchy reset line on your motherboard, you can cover the reset pin (B2) on the ISA connector with an insulating tape (Kapton would be [...]

I gave this a try but so far no better and no worse. No luck with my 440bx so far either so there must be something off with my build. I will keep trying with the 440bx though. 

Do you have any estimate on when you might release the next design? I was considering picking up another set of components but I will probably wait for next revision instead.

skadarnold, 2023-11-04:
I am following the vogons thread now so I can see your plans for the next release. I will move this conversation there as well. I think I can run some debug through the UART and see if I can get some more info from there.

skadarnold, 2023-11-04:
I just made an interesting discovery that may or may not help narrow some things down. When I load the pg-mpu.uf2 it works every time 100%. Anything that initializes the DAC seems to be hit or miss.

skadarnold, 2023-11-06:
I am going to close this since it does not seem to be a short to ground or an issue with the DAC module.

closed 2023-11-06 by skadarnold