polpo/picogus #23: GUS Emulation - Sound Effects in some games, but no Music. Other games crash

mikboy018, 2023-11-28:
So I ordered a couple of the 2.0 models, and put one in a 486, with another in a Pentium 2. They're both giving me similar interesting results, in that I can't get any GUS music to work. I'm pretty sure the hardware is fine, as MPU-401 worked great for Day of the Tentacle, X-Com, and Loom on my old 486. The ultimate talkie edition of monkey island 1 had some problematic music, but since that isn't really official, I don't think its as big a deal as GUS Music for me.

For GUS, I've tried Rise of the Triad (Shareware and Dark War), Raptor, Tyrian, and Jazz Jackrabbit (+ both Holiday Hare 94 and 95). 

For Rise of the Triad:
- Choosing Ultrasound for SFX appears to work, although I get no audio
- Choosing Ultrasound for Music crashes

For Raptor (full and shareware):
- Running setup causes no issues. On the Pentium 2, running the full game crashes on Music initialization, while on the 486 it launches, with no music or SFX.

For Tyrian:
- Running ultramid prior to, launching setup, and then choosing Gravis-MIDI causes a runtime error (I can get a screenshot if it's helpful)

For Jazz Jackrabbit (HH94/HH95/CD):
- Setup indicates it picks up the ULTRASND settings ('set ultrasnd=240,1,1,5,5' used, but I've also tried 3 for DMA on both systems).
- Launching the game gives no errors, and sound effects play, but music will not.

I don't know the chipset of the 486 off-hand, but the Pentium 2 is a a socket 1 / ASUS P2B-F (intel 440lx chipset). 

Anything else I should try/you'd like to know? I'm pretty new to GUS, so I'm not sure if I missed a step. I did run on both with the latest picogus release and the ultrasnd drivers/patches from the wiki.

mikboy018, 2023-11-28:
Of note, this was the first time my 486 ran Day of the Tentacle without softMPU for my MT-32... I used to get memory errors, but that is no longer an issue!

deepthaw, 2023-11-28:
What version of Tyrian? I've been able to run Tyrian 2000 pretty well perfectly using ULTRAMID and the Gravis MIDI option in game.

mikboy018, 2023-11-28:
Shareware (from an old demo disk I had as a kid) and Tyrian 2000 (from GoG)

mikboy018, 2023-11-29:
Thanks!

Ran both cmod & gusdram on the Pentium 2, as well as the 486.

In both cases, cmod froze, and gusdram reported no issues (photos attached).

[image]
[image]

mikboy018, 2023-11-29:
Ok update for the 486: working great.

I pulled the Aztech Sound Galaxy Washington out, and audio just worked in Jazz Jackrabbit. It's jumper was set to 220, and my ultrasnd variable was set to 240,3,3,5,5 as to not conflict.

[image]

The Pentium 2 also has a 2nd soundcard, a Creative SB32... think it also conflicts?

polpo, 2023-11-29:
Your autoexec.bat looks OK to me - doesn’t look like anything should conflict. This is feeling like a DMA issue to me given the behavior you’ve described so far. Thanks for all of the detail you’ve given, it’s really helpful!

I have a board with a 440LX chipset and an AWE64 which should be pretty similar to your SB32… I’ll try to see if I can replicate the issue with that and the PicoGUS in the same system.

mikboy018, 2023-11-29:
Nice! Looking forward to seeing what you find on your end.

I did identify my 486 motherboard... [link here](https://theretroweb.com/motherboards/s/unknown-isa-s486) (well, it's 'unidentified' here, but at least it has some detail)

mikboy018, 2023-12-01:
Forward progress!

Raptor (Shareware) is working on the P2. I was using Phil's Computer Lab's MS-DOS mode made easy, and had been updating both autoexec.bat for windows and for the *.pif he created. What I didn't look at was Windows Device Manager, which set the SB32 to IRQ 5 anyway...

I disabled PNP OS in the Bios, Manually set my IRQ/DMA, and it's working now!

mikboy018, 2023-12-01:
I don't know a lot about GUS's, but should this be able to do music and sfx at the same time?

deepthaw, 2023-12-01:
I think most games that support GUS can do sound and music simultaneously on it. It’s all the same thing to the GUS so it can’t even tell the difference.

mikboy018, 2023-12-01:
I'm good to close this, unless we want to try some things to see about compatibility with that symphony chip... happy to try stuff/test that as needed.

I mainly have it in the 486 to serve as an MPU-401 card, and it does that well, looking through my launch settings, even monkey island 1 (ultimate talkie) is working.

In any case, thanks for the help!

polpo, 2023-12-01:
I'm glad you've sorted things out! I haven't really used Win9x's MS-DOS mode so I hadn't thought about how that'd change your SB32's settings.

An IRQ conflict definitely explains the issues - for example in Jazz Jackrabbit, the GUS's timer IRQ drives the music playback while sound effects are triggered by the game independently of the music playback routine.

How are things on the 486? Is there still a conflict when you have the Aztech card installed?

mikboy018, 2023-12-01:
There are still conflicts on the 486. It wasn't one of the 'compatible' chipsets, but it seems like its the exact same issue. My autoexec.bat settings are set to that picture I took above.

I could try different IRQ/DMA combinations to see what works. That system doesn't have much in the way of BIOS settings I can modify.

mikboy018, 2023-12-03:
ICU worked to a degree.

I can't disable LPT from the BIOS like I could with the 486, so instead of setting the IRQ to 5 or 7, I went with ICU's suggested default 9(2), and Jazz Jackrabbit worked perfectly.

I went to try Raptor and Tyrian, and Raptor gave no sound. For Tyrian, I didn't attempt, as ULTRAMID threw the following error:
```
configuration file not found in ${ULTRADIR}/midi directory
```
Going to do a fresh reinstall of everything in C:\ultrasnd tomorrow, as  that path is set right (autoexec.bat below)
[image]

cmodtest also hangs on loading strshine.s3m

mikboy018, 2023-12-03:
No luck yet with Ultramid.

Disabling the Aztech (Sound Blaster Compatible) within ICU, & updating its IRQ to 7 in Autoexec.bat allows SCUMM games to use my Aztech for voice, and the picoGUS as an MPU-401 -- which was my real intended use case on this system. It's a bonus I can play Jazz Jackrabbit on it, especially this time of year.

Games working great:
picoGUS only: Jazz/HH94/HH95 (GoG Copy)
picoGUS (MPU-401): X-Com
picoGUS + Aztech: Any SCUMM games (including talkies)
Aztech only: Halloween Harry

Games not yet working:
Raptor (No sound at all from picoGUS, sfx from Aztech does work -- but not music)
Tyrian (Ultramid not yet working)

Honestly, I'd play raptor/tyrian on the Pentium 2 anyway, so I'm pretty happy with how this is going. Both cards are working great, but I definitely have some resource conflicts to figure out on the old 486.

B24Fox, 2025-10-04:
> No luck yet with Ultramid.
>
> Disabling the Aztech (Sound Blaster Compatible) within ICU, & updating its IRQ to 7 in Autoexec.bat allows SCUMM games to use my Aztech for voice, and the picoGUS as an [...]

Hi!
Did you by any chance got **Raptor** to work?? 

I have a SuperSocket7 board (VIA MVP3 chipset), and I have the same issue in GUS mode. When selecting "UltraSound" or "UltraSound Max" in the game's setup. "Music" makes it freeze on game launch; and "Sound FX" just give no sound.

Raptor music when selecting MIDI (i have a daughter-board plugged in) works fine though, in both GUS & MPU-401 modes. 

The behavior is the same in both Win98SE & DOS mode.