polpo/picogus #31: Any chance to bring MIDI Input support?

Deftaudio, 2024-02-08:
Hi, great work and an amazing project!
Is there any chance to bring MPU-401 MIDI Input support in firmware?
Hopefully you have an extra pin for that. I can use an external optocoupler. Just wondering if it's possible at all in the current design. Thank you!

Deftaudio, 2024-04-11:
Sorry for pinging here again, any update please?

polpo, 2024-04-11:
Unfortunately with the current design it's not possible to support a MIDI input as there are no free pins that can be used for it. There's a UART TX pin exposed on the board, also used for MIDI output but can't be remapped as an RX pin. It may be possible to free up a pin in a future revision and it's something I'll keep in mind.

Deftaudio, 2024-04-11:
Thank you Ian, I understand that you might ran out of available pins. Can you add support for a USB MIDI class compliant device then? This way any external USB MIDI interface will work just fine.

polpo, 2024-04-11:
Oh, that's a really cool idea! TinyUSB can support USB MIDI class devices with this driver: https://github.com/rppicomidi/usb_midi_host
I'll look and see if such a thing would be possible.

closed 2024-04-11 by polpo

kmeaw, 2024-07-03:
Just in case there will be some plans on throwing in some mux to free pins for MIDI input, please consider also having the wavetable header handled separately from the MIDI output - so software could switch between handing over the MIDI data either way (or both, as current PCB does), giving the user choice over the routing.