polpo/picogus #36: Can't program the PicoGUS using Amiga Bridgeboard PC

jfharden, 2024-03-18:
I'm using the PicoGUS in an Amiga 2000 using the A2286 bridgeboard (which is a full IBM AT class PC on a card, an 80286 at 8 MHz), I also have an XTIDE, and a VGA card on the ISA bus, but nothing else.

Prior to 1.0.2 the GUS sound didn't work, but I think fixing the timings for the 5170 has also fixed it for this. GUS sound seems to work perfectly now, as does adlib (I've tried Dune 2: The battle for Arrakis and Planet X-3 for adlib, and I've tried the Legend demo and The Good, Bad, and Ugly demos for GUS sound).

However programming the Gus with `pgusinit /f <file>` doesn't work. It prints a lot of garbled characters. I don't have a way to copy and paste them but you can see in the screenshot attached the garbled errors.

Even after the output in the screenshot the GUS still works, so I don't think it's actually managing to program it at all.

[image]

If there's anything I can do to help debug this let me know and I will do my very best. It's absolutely incredible to me that this works at all in the Amiga

jfharden, 2024-04-24:
As of version 1.2.0 (I haven't tried 1.1.0) this doesn't happen anymore, occasionally flashing the firmware from pgusinit works, but mostly it hangs after outputting a number of .'s

On one occasion it threw an error saying the picogus wasn't in programming mode and returned to the DOS prompt, but mostly it hangs forever.

jfharden, 2024-04-24:
As a slight aside, the new 1.2.0 soundblaster firmware is working great otherwise with this amiga bridgeboard! I tried planet x3 for adlib mode which works great, and modmxt and played some mod files using soundblaster output.

This is incredible! What an upgrade :bow: Thank you to all who contributed to it!

jfharden, 2024-04-24:
I just realised I haven't mentioned at all that I'm running DOS 3.3, not 6.2 or above as I imagine most people are.

polpo, 2024-04-24:
Thanks for the reports! DOS 3.3 shouldn't be an issue. There have been some fixes over the past few firmware releases to improve bus reaction time and reduce the amount of time IOCHRDY is held, but the chipset used on the Bridgeboard must be especially sensitive. Unfortunately I don't have access to an Amiga with a Bridgeboard so I can't know for sure! But if I do anything to change bus timing in future firmware releases, it will be in the release notes. Good luck and it's great to hear it's working well otherwise!