polpo/picogus #37: Crackles and clicks when playing music

Piligrim-77, 2024-04-06:
Hi,
I have assembled three PicoGUS and two of them work fine, and on the third board, when playing music, you can hear clicks and crackles like when listening to a vinyl record. 
The problem is observed in the XTC-PLAY player. 
The sound is normal in games. 
What could be the problem? Is this a defective PCM5100A audio codec? 
Link to the recorded audio file:
[https://drive.google.com/file/d/13oy0wXh3Ue7P_ZVYZhUcEBK353x5l1JZ/view?usp=sharing]

nikitalita, 2024-04-21:
There is apparently a known issue with around 10 percent of PCM5100A chips where they refuse to play 21.05khz audio. If this is an intermittent issue that only affects certain applications (particularly when mixing multiple channels), then this is likely it. The solution would be to replace the PCM5100A.

Piligrim-77, 2024-04-21:
I think the problem is not with the PCM5100A. I bought 5 pieces of new chips and replaced two of them. Nothing has changed, the problem remains. 
I also tried other programs and games - clicks are also heard in them. 
Could it be buffer chips or something else?

polpo, 2024-04-21:
Is the issue audible in non-GUS modes? If it’s only on GUS that points to faulty PSRAM chip (U8) or a soldering issue around that chip. It’s not used in the other sound modes.

Piligrim-77, 2024-04-22:
I checked other modes - when emulating SB, the game freezes, the music is heard for less than a second, then a click and silence. The only mode that works fine is MPU 401 emulation.