polpo/picogus #43: Combined SB + MPU

eivindbohler, 2024-05-01:
I'm apologize if this has been asked before, but would it be theoretically possible to combine the SB and MPU firmwares/functionality?
I'm very impressed by how this project is progressing, and I'm considering trying to integrate it into another revision of my x86 SBC, [the TinyLlama](https://github.com/eivindbohler/tinyllama2), possibly ditching the Crystal CS4237B altogether. I guess using two rp2040s each running its own firmware could be doable as well, just wanted to check! :)

skadarnold, 2024-05-01:
Having MPU combined with other firmware seems to be a highly requested feature, and makes a lot of sense. I am almost wondering if the MPU can start to function like the /j option that can be enabled on anything. I would even appreciate having GUS+MPU too since I have found GUS audio sounds better than SB on games that support both, but might want to use MPU for music still. They are rare cases, but having all these options like this are opening up new possibilities.

polpo, 2024-05-01:
In the latest firmware release, I eliminated all busy waits in the MPU-401 code in anticipation of combining it with the other modes. I'm wrapped up in other things but It's definitely something that is possible and want to do in a release soon.

skadarnold, 2024-05-01:
> In the latest firmware release, I eliminated all busy waits in the MPU-401 code in anticipation of combining it with the other modes. I'm wrapped up in other things but It's definitely something that is possible and want to do in a release soon.

Super cool! Seems like things are moving at 1000mph right now so totally understandable. Everything everyone is contributing at this moment is complete gold so it is totally worth any waiting. Thanks again!

eivindbohler, 2024-05-01:
That's fantastic, @polpo! Again, amazing work! 🙌

kmeaw, 2024-07-17:
I understand that the feature isn't complete yet but I still tried e0ef9c2d3 with a minor fix (send_midi_byte -> send_midi_bytes) - it boots fine and SB is detected but any attempt to play PCM audio or music causes the card to emit some high-pitched slow random noise.

polpo, 2024-07-22:
@kmeaw that branch has some work I made to get stereo SBPro working and I pushed the commits the night before before leaving on vacation so I wouldn't lose any work. It's totally untested; don't worry, it'll get fixed up before I release. Next release will be targeting SBPro 1.0 and MPU simultaneous with other modes.

polpo, 2024-07-30:
This is released in firmare v2.1.0: https://github.com/polpo/picogus/releases/tag/v2.1.0

closed 2024-07-30 by polpo