polpo/picogus #46: SB opl2 emulation missing certain notes

TerryFi, 2024-06-05:
In Prince of Persia, the intro music missing some piano notes. Also when games starts "Danger" music piece missing a lot of notes in the piano chord.
Issue also present in other emu8905 project,  verified by playing VGM of the game in emu8905 based VGM player.  
Narrowed it down to a bug in attack rate calculation of emu8950.

perchrh, 2024-06-05:
What's this emu8905 you refer to? I tried searching for it but couldn't find any matches.

polpo, 2024-06-05:
> What's this emu8905 you refer to? I tried searching for it but couldn't find any matches.

emu8950 is the OPL2 emulator used in PicoGUS. I use it because it was optimized for the rp2040-doom project so it can run on the RP2040.

polpo, 2024-07-08:
Fixed by #47 and #52. Thanks again @TerryFi!

closed 2024-07-08 by polpo