polpo/picogus #48: Where am I going wrong?

Pontiac76, 2024-06-06:
I bought a PicoGUS a couple weeks ago, and just tonight I'm trying to get this thing rolling.

System Stats:
486DX2-80
Trident 512k VGA Card
XT-IDE Board, configured at $D000 and $360, with an IDE2CF drive plugged in, 2Gig CF running MSDOS 6.22. 
PicoGUS set to IRQ 5, DMA1, just as it was right out of the box.  (Twin jumpers on DMA1)
Additional card for serial, IDE and FD, but, IDE is disabled via jumpers.  This card is set to run at $C800.
100mbit D-Link DE (Or DF? Through way right at the bottom) 220T NIC, running at packet int 0x62, 0x300, IRQ3.

Software:
- Downloaded the 1.2 package today, and extracted it to the CF
- Downloaded the GUS software and extracted it to the CF

What I've seen since the initial plugin:
- The yellow blinkie light shows up, and stays steady on when plugged into the motherboard.

Actions I've taken:
- Plugged the GUS into my Win10 PC via USB, eventually found the RPI drive, initially copied the SB UF2 over (Since I want minimal SB)
- After dragging the UF2 over, it seemed to have rebooted, noticed the LED on the GUS light up, drive mount went away.
- Plugged the GUS into one of the 3 spare ISA slots, pgusinit didn't see a card.
- Shut down the machine, moved the card to another slot, reran pgusinit, again no card.
- Found instructions about setting the environment.  Wrote a batch file to set the vars.  Tried 220 and 240, still no go.
- Plugged the GUS back into Win10 and put the pg-gus.uf2 on
- Same results.  LED turns on, drive goes away, plug it into another random ISA, still can't find a card.
- **_Went minimalistic_**; Pulled all cards except video, the PicoGUS and the Floppy controller.
- Booted to MSDOS 6.22 with no drivers
- Switched to the v1.2.0 PicoGUS disk
- Ran pgusinit, no parameters, still found no card.
- For fun, entered the environment variable to set the address, and still no go (With the GUS UF2 installed)

I'm at a loss on where to go from here.

The card "seems" OK since I can put the firmware on the drive via USB and things do their things, but I'm not sure where to go from here now.

Pontiac76, 2024-06-06:
I think I found a regression.

I went to v1.1.0 and run the pgusinit and I got garbage back.

Pontiac76, 2024-06-07:
I tried going back a version to 1.1.0.  I see that PicoGUSinit is 2.0.1 versus 2.1 in 1.2.  I pulled the card out and plugged it back into my Win10 machine, put the v1.1.0 GUS firmware on, it did its spiel, plugged it back into the 486 and I still can't see the card.

polpo, 2024-06-07:
What is the ISA speed set to, if your BIOS gives you the ability to change it? Because you have a 40MHz FSB that may mean by default it's running ISA at a higher speed than normal. And what chipset does your motherboard have?

This is a bit of a tangent, and while it wouldn't prevent the card from being detected at all like what you're seeing, one thing to watch out for is that the PicoGUS in GUS mode (and real GUS) don't play very well with the XTIDE hardware, for reasons still unknown to me. It does work fine with the XTIDE Universal BIOS, though, so if your motherboard's BIOS won't recognize your CF card, I'd recommend using XUB along with the IDE on your super-IO card, and you'll get better performance that way as well.

Pontiac76, 2024-06-07:
https://media.discordapp.net/attachments/762944299767037972/1248706208957730936/20240607_143003.jpg?ex=6664a3c8&is=66635248&hm=f934cd961e208ae1f8c6bec5c8abf8f1e936f74953cac9c2d5263e853d0baec1&=

That's the closest thing I could find to setting the ISA speeds.  I changed the ISA Bus Refresh to Slow and brought it from 1/1 to 1/64, but still have the same results.

As for the chipset, there's nothing on the bootup screen to indicate anything, so here's a pic of the mobo.  (I removed the floppy cable for the shot)

https://cdn.discordapp.com/attachments/762944299767037972/1248708132989833399/20240607_143903.jpg?ex=6664a592&is=66635412&hm=087137f75be3d31d6b3faddc21d44e345c2aaf9e2371e0f1e01b68bdd5467c66&

A940243RD is all I have that's questionably readable in the photo.

Checking the manual for this board (Closest i could find) https://stason.org/TULARC/pc/motherboards/A/AMPTRON-INTERNATIONAL-INC-486-DX-6900-VER-1-75.html

I pulled JP16 (Bottom of the page) that says if I pulled JP16, it'd take the VL Bus Speed down to under 33mhz, still no answer.  I then pulled what I think is JP17 (Not labelled but on the same header as JP16) and again no change.  I've been using 1.2.0 for this specific test.

Oh, one other hardware thing is that I am not using a modern PSU.  It's the original PSU I received with the case, which is full AT mode.  It does have a -5v line.  I did have a modern supply with an adapter cable, but found out that some cards require the -5V line, so been risking it.

Pontiac76, 2024-06-07:
Actually, I just found a setting that didn't look like it was directly related to the ISA bus...  

https://cdn.discordapp.com/attachments/762944299767037972/1248716921499811861/rn_image_picker_lib_temp_38e64db8-fa71-43f4-a5e9-59348254b450.jpg?ex=6664adc2&is=66635c42&hm=fcaa32cc52c9807947ba46f432a0da495a6186c2ad533f31ca9913381246b611&

The PCLK/2 was the culprit.  If I switched it to /3, it works.  I may be a bit unstable now with the 2-1-1-1 WS, but I'll deal with that later.  (I didn't think to check the settings on that before I made changes)

I found this out by going back to basics and trying to set everything to as "stock" as possible and let the computer decide what it should be.  When I set the Auto Config to "Enabled" the GUS started to work.  So I started playing with the Cache Read Hit Wait States, still to no avail.  I then went down to the AT Clock Speed, hit to /5 initially, and it worked!  So switched to /3 to work "up" from 2, and this time it worked.  I got a proper answer, noting that the Adlib firmware was installed, which is correct from 1.1.0 when using the 1.2.0 version of pgusinit.

Thanks for the hint on this.  At least now I know I don't have a defective card.

Next step is to put the rest of the hardware in and see if I can get things rolling.

I HAVE heard rumors about the XTIDE not being happy about other things.  But at a minimum specification I know things are working, so I can go up from there.

Thanks again!

closed 2024-06-07 by Pontiac76