polpo/picogus #49: Joystick input affected by GUS utilization

ehoughland, 2024-06-08:
In certain games, the joystick input is affected by the utilization of the soundcard. For example, in Super Street Fighter 2 Turbo, if GUS firmware is loaded with joystick support turned on (/j) AND the card is used by the game for music and sfx, the joystick will exhibit random inputs. If the game instead is directed to utilize another soundcard within the system for music and sfx (in this case, an ESS Audiodrive), the joystick behaves normally.

polpo, 2024-06-08:
Hi, I have also experienced this on the PicoGUS in GUS mode with any game that uses DMA for streaming audio. I've also experienced it on a real GUS with a real gameport joystick. On the real GUS it's not as bad but it still can make the game unplayable. I don't really think there's a way to really fix this, unfortunately. I'll keep this issue open until I create a "Known Issues" wiki page.

ehoughland, 2024-06-08:
Interesting, thanks for the reply. Any chance six button support or remapping is coming? 🤞