polpo/picogus #50: Adding CDROM emulation support?

toddfraser88, 2024-06-08:
I know this is probably a long shot but I was wondering if you could get in touch with Kevin Moonlight and ask him if he can share his Panasonic CDROM emulation code for use with the PicoGUS?

He made a video about what he's done here -

https://www.youtube.com/watch?v=L59Bp61Je4Y

At present we don't really have a way of getting CDROM audio working in DOS on real hardware without a real CDROM drive, so working CDROM emulation would be nice to have.

polpo, 2024-06-08:
Hi, I’m in regular contact with Kevin. Being able to use his CD-ROM emulation is a definite possibility for the PicoGUS. Stay tuned!

toddfraser88, 2024-06-08:
Will do!

Pontiac76, 2024-06-09:
... can... you have that done by Thursday last week? {smirk}  I've been LOOKING at another solution, but, if there's enough room on this PGUS.... Wow.

polpo, 2024-06-09:
The big hurdle is getting USB mass storage fast enough. As is with the TinyUSB library, it's not fast enough to emulate a 1x CDROM drive. AFAIK it's a software issue, not a hardware one.

polpo, 2025-05-18:
Thanks to the efforts of @rppicomidi and @wbcbz7, USB mass storage is now fast enough to support emulation of a double speed CD-ROM drive. And thanks to @yyzkevin, he has reverse engineered the Panasonic drive interface.  Look for CD-ROM support in a future firmware release!

RetroLoom, 2025-05-18:
> Thanks to the efforts of [@rppicomidi](https://github.com/rppicomidi) and [@wbcbz7](https://github.com/wbcbz7), USB mass storage is now fast enough to support emulation of a double speed CD-ROM [...]

This is amazing news! Will this also include redbook audio streaming support or data only?