polpo/picogus #53: Can't init femto with ISA adapter

kp73, 2024-06-26:
Hi, trying to init and get an error;

`ERROR: PicoGUS card using protocal 221, needs 2`

Using a femto with ISA adapter from Tindie https://www.tindie.com/products/polpo/picogus-femto-edition-for-pocket-386-and-isa/

Programmed the Pico by holding BOOTSEL and dropping pg-gus.uf2 from picogus-v1.2.0 release (unplugged from femto).  

Stuck it in my 486 using the ISA adapter, but not joy.  Will try it in the book 8088 when I find the power adapter :)

A standard PicoGUS v2 works fine in the 486

[image]

kp73, 2024-06-26:
Seems to work OK on the book808 

[image]

Any idea why it won't work with the ISA adapter?

polpo, 2024-07-08:
Hi, sorry for the delay in getting back to you. There are probably solder bridges on the ISA adapter on the fine pitched pin connector that the Femto plugs into. Take a close look, if you are handy with soldering you can clean any bridges up with some flux or solder braid. If not I can send you a new one - send me a message on Tindie and I'll take care of you.

closed 2024-07-08 by polpo