polpo/picogus #54: Compatibility Report: 5150 and 5160

16k, 2024-07-06:
I tried the latest picogus 2 which I obtained from polpo earlier this year in my 5150 an 5160 motherboards.

On the 5150 as half expected it was not detected at all.
- 64kb-256kb motherboard
- ROM: 10/27/82
- RAM fully populated to 256kb
- Other cards: PEGA1A EGA card, Quad flop floppy adapter with no floppy bios. (only EGA ROM)
- Soundblaster firmware installed on picogus (v1.2)
- Booted with 360kb disk and ran pgusinit.exe (v1.2) -> no picogus detected.

On the 5160 it seems to work fine
- 256kb-640kb motherboard
- ROM: 05/09/86
- RAM fully populated to 640kb
- Other cards: CGA, adaptec scsi jumpered to floppy only mode. (no ROMS)
- Soundblaster firmware installed on picogus (v1.2)
- Booted with 360kb disk and ran pgusinit.exe (v1.2) -> picogus detected no problems
- I've also previously tested this motherboard a few months ago
   in a more complete configuration with an xtide and various games
   seemed to work fine in the old version's adlib mode (haven't tested anything more fancy)

Thanks for an awesome product!

polpo, 2024-08-22:
Thank you for this report! I have tested PicoGUS on a 5150 somewhat recently and it did work. I'll leave this report open until I can verify if there is a regression.

williami-alchemy, 2025-05-11:
I'll be testing this again in the coming weeks, with the latest releases.

16k, 2025-05-11:
I'm going to test this again with the latest release sometime in the coming weeks.

jtgans, 2025-10-06:
Just confirmed detection does *not* work with version 3.2.1 on a picogus 2.0 in the slot furthest left on a 64KB 5150 with the latest BIOS