polpo/picogus #55: Add support for legacy Gameport controllers?

kikendo, 2024-07-30:
I was wondering if this was possible, maybe via this device:
https://github.com/necroware/gameport-adapter

The main reason I haven't dumped my SB16 in favor of the PicoGUS on my main setup is that I love to use my Gravis Gamepad. Would love to be able to use it with my PicoGUS!

AhmedAbdulFaisal, 2024-08-07:
I would suggest getting a system working where you can use both sound cards in the same machine. Easier done than said in some cases, as long as you aren't doing anything like using IRQ7 for printing and you make a physical note of the resources of both sound cards. And of course, this is what GUS users did back in the day.

kikendo, 2024-08-07:
The reason I suggested this is because I am running PicoGUS in parallel with my SB16 and it’s a freaking nightmare of configuration and conflicts. 

Also since now you can easily switch emulation modes on the fly, I feel the PicoGUS is well suited to be the only card in a system. 

There’s s already support for some controllers. I think classic ones should be supported too. Using a Dualshock with an old system is the wrong vibe 🙃