polpo/picogus #66: GUS Compatibility regression in v2.1.0 & existing issues

vermian, 2024-11-11:
PicoGUS v2.0 Setup:
Port 240, IRQ 5, DMA 3
Firmware v2.1.0

Using MS-DOS v7.1

GUS Issues:
-Jazz Jackrabbit CD - Music doesn't play at all with firmware v2.1.0. Sound still works fine. Everything including music was working fine with firmware v1.2.0.
-Tyrian 2000 (v3.0) - Music is all weird with firmware v2.1.0 and notes are stuck after you leave setup. Comparing with v1.2.0, Ultramid didn't even detect the GUS. 
-Minebombers (v3.11) (1996) - Music doesn't play at all, but sound effects work fine. Same issue with both firmware v2.1.0 and v1.2.0
-X-COM: UFO Defense v1.4 - Firmware v2.1.0: music works, but sounds don't play at all. Game v1.2 - sound is just random instrument noises? (GUS music was not implemented in game version v1.2 yet AFAIK)

polpo, 2024-11-13:
Hi, thanks for the report. Music not playing but sound effects being OK points to issues with the IRQ - is IRQ 5 shared with any other cards in the system? Of the titles you mention I've only tested Jazz Jackrabbit in depth on v2.1.0 firmware (it ran all weekend at VCFMW with no problem) but I will look at the others. I don't have a 440ZX system but I have a 440EX and 440BX that I can test on.

vermian, 2024-11-13:
I have a Yamaha Audician 32+ but it's set to Port 220, IRQ 7, DMA 1. If it was an IRQ issue, wouldn't it also affect the older firmware?

polpo, 2024-11-23:
@vermian - I've made a special branch that builds the GUS firmware without MPU support as that's the most likely cause of the regressions you're seeing. Can you download the "PicoGUS Firmwares" artifact at the bottom of this page: https://github.com/polpo/picogus/actions/runs/11989912827 and try it out to see if it improves compatibility for you?

vermian, 2024-11-26:
> @vermian - I've made a special branch that builds the GUS firmware without MPU support as that's the most likely cause of the regressions you're seeing. Can you download the "PicoGUS Firmwares" [...]

Yeah, looks like that fixes the Jazz Jackrabbit music issue. So both music and sound work fine again.

Other differences:
-Minebombers (v3.11) (1996) - frozen black screen on game start. Before I at least had sound effects.
-X-COM: UFO Defense v1.4 - Both sound and music work now, though music seems to have stuttering in the intro sequence and during the main menu. I didn't hear stuttering music in-game but I only played for a couple of minutes.
-Tyrian 2000 (v3.0) - not sure if it sounds different because I only listened to the music a short while before but maybe it sounds fine now, but I still got a frozen note during the game on the main menu after a demo sequence.

vermian, 2025-06-29:
Tried newest v3.1.0 firmware and Jazz Jackrabbit, Epic Pinball, Minebombers have no music but do have sound, similar to v2.1.0.

p-rintz, 2025-07-16:
Funny enough I have music in Jazz Jackrabbit, but no music in Duke 3D or Descent. (Sound effects work everywhere).