polpo/picogus #69: “Building your PicoGUS” wiki page full of errors

JoopIdema, 2024-12-15:
The “Building your PicoGUS” wiki page states it describes buiding a v1.2 PicoGus.

But the link to the BOM.csv does not work and the link to the Mouser Projects list the components needed for the v1.1 PicoGUS. I did not notice this so I got several wrong and missing components resulting in placing extra order and thus costs.

Also the link to the interactive BOM is for the v1.1 version, so useless when building a v1.2

Furthermore the BOM (DAC module) v1.2 found hw folder lists R7,R8 to be mounted but when using a DAC, R8 and R7 do not need to be mounted!

Also adding a picture of a complete board might help when building it.

All and all, for me, it let to a very frustrating project to build.

korneliuszo, 2024-12-15:
https://github.com/polpo/picogus/blob/main/README.md#license
https://spdx.org/licenses/CERN-OHL-P-2.0.html

karoliskap, 2024-12-29:
@JoopIdema can you post what exact components are missing or wrong? I about to order parts from Digikey, so it would be very helpful.

JoopIdema, 2024-12-30:
I had to order U2 and U10 extra. And some caps I believe.
You can find the BOM file for the 1.2 version of the PCB in https://github.com/polpo/picogus/blob/main/hw/BOM%20(DAC%20module).csv

oXL0C0Xo, 2025-02-23:
Just looking this over as I plan to build a card that uses the DAC module. It was mentioned that R& & R8 do not be mounted when using a DAC module by the OP; is that the case? The reason I ask is because they are still listed in the BOM in the project folder.

crtified-retro, 2025-03-25:
Sorry that the build guide has become a little out of date. I've long meant to contribute some updates, but life is busy. Seeing this discussion gives me impetus to get to work. I'll try to get some updates written up and proposed within the next month.

> Just looking this over as I plan to build a card that uses the DAC module. It was mentioned that R7 & R8 do not be mounted when using a DAC module by the OP; is that the case? The reason I ask is because they are still listed in the BOM in the project folder.

R7 and R8 are not meant to be in the BOM for the DAC module version of the build. So leaving them out, as you have done, is exactly right.