polpo/picogus #74: USB gamepad emulation not working PicoGUS 2.0 on firmware 2.2

nyandeyanen, 2025-03-09:
I've been attempting to get a gamepad working in DOS on the PicoGUS under firmware 2.2 for a while, I've bought several gamepads to try with the PicoGUS' USB port including the following:

Logitech F310 Gamepad (switch set to Xinput mode)
8bitDo Utimate Wired C Controller for Xbox
Sony DualShock 4

I saw the DualShock 4 listed in the hidd_app.c in particular, but so far, well, no joy. Enabling the joystick with /joy 1 shows that pgusinit enables the port, but none of the listed gamepads are detected. It doesn't seem to be dependent on mode, as sb, gus, tandy and mpu401 modes all seem to fail. In each case going into any applicable game setup program, for example setup.exe in Wacky Wheels, indicates that there's no joystick attached to the computer. I've tried this with the onboard sound (ESS Solo-1) disabled in the BIOS, and with the Solo-1 enabled but with its gameport disabled, and in all cases no joystick is found on the port. The DualShock 4 in particular displays an orange light in the back of the controller where typically it displays blue light when plugged into my "modern" computer (Debian Linux).

My system configuration:

Biostar M6TEA / Intel 440EX chipset with a Celeron 266
Windows 98SE set to boot into DOS mode directly by disabling BootGUI in msdos.sys
Either an EMS or XMS configuration with EMM386 and himem.sys doesn't seem to make a difference, neither does using JEMM
A Dreamblaster is attached to the PicoGUS' waveblaster header

FWIW, the sound modes all work quite well, perhaps some pitch glitches in Tandy mode (Castlevania in particular) but otherwise I'm very happy with the card. I've even used PicoGUS' MPU mode with the ESS Solo-1's SB emulation for the Dreamblaster by disabling the MPU port on the Solo-1. This is truly an outstanding piece of hardware, well worth putting together a system with ISA slots. :)

nyandeyanen, 2025-03-17:
I've continued attempting to get the gamepad support working since last week, using the DOS Thrust program by Bret Johnson I'm able to monitor what the system sees when the /joy flag is set to either 0 or 1. When the PicoGUS is set to joystick disabled, the thrust program shows all 4 buttons (1 and 2 for both joysticks) as being missing rather than "Off". When the PicoGUS is set to joystick is enabled, it reports the buttons as being constantly set to "On"/pressed. Neither mode reads out any axis data, not surprising when it's disabled, perhaps somewhat surprising when it's enabled. The PicoGUS is in usb mode to avoid any possible issues with sound modes.

I've ordered a PicoGUS 1.2 board kit and a Pico W to see if I can reproduce the problem on a different board, hopefully I'll get better results once that comes in. I'll also be looking into the picogus source to hopefully understand what's happening here better.

polpo, 2025-03-17:
Hi, sorry you're having problems with the joystick emulation. It should work fine on the PicoGUS 2.0 and firmware 2.2.0 (there was just [a video on the ctrl-alt-rees YouTube channel showing it working](https://www.youtube.com/watch?v=z4NYW5GQ3vc)), but it is kind of picky about what joysticks it supports. For example, I have a Logitech F310 and cannot for the life of me get it to work. I can't speak for how the 8bitdo works, but the DualShock 4 should be fine. I can test DS4 support again just to make sure it should be OK.

Just to be sure – it looks like the Biostar M6TEA has onboard audio and game port. Is the onboard game port disabled? Unfortunately game port emulation cannot coexist with another game port in the system.

In general, gamepad emulation needs a big overhaul, not only to support more joysticks but provide better troubleshooting support. For example, helping people answer the questions "does the PicoGUS even see my joypad?" "are there other game ports in the system?" would help a lot.

nyandeyanen, 2025-03-17:
Hi,

Thank you for your response, and for this amazing card! The gamepad support is disabled on the M6TEA's onboard ESS Solo-1 sound chip by editing the C:\ESSOLO.INI file in a hex editor and changing the gamepad address field to an invalid address prior to running the essolo.com DOS driver. The driver reports that the gamepad port is disabled along with displaying the address/dma/irq/MPU-401 information. I set the ESSOLO.INI file to readonly afterwards to avoid the Windows driver making changes to the configuration.

When attached to the system's USB port via a hub in this configuration I was able to get the Logitech F310 working in Dinput mode under DOS using the USBDOS suite, loading the USBUHCI, USBHUB and USBJSTIK drivers while the PicoGUS' joystick port is disabled. Notably the DualShock 4 doesn't work in this configuration, but it's possible to use port 201h with the F310 from both the Thrust utility and Commander Keen 4 - although not for some other games. The DualShock 4 does show up in Windows 98's "Gaming Options" control panel as a "Wireless Controller", so it can work on the system via USB, just not on the PicoGUS' USB port. FWIW I have a Logitech Wingman Precision connected to the motherboard's gamepad port which does work in DOS when the motherboard's built-in gamepad support is enabled on port 201h in the ESSOLO.INI file.

I've started to think that perhaps the PicoGUS 2.0 card I received might have a hardware issue, although everything seems to be working except for its USB port. The 1.2 kit should be arriving tomorrow, once I have time to assemble it I'll see if I get different results with that. If the 1.2 board works maybe I'll go poking around the 2.0 card's USB port with a multimeter looking for bad traces or solder joints.

Best,
Les Barrows

nyandeyanen, 2025-03-22:
The 1.2 board arrived a bit later than expected, and I noticed that of course it doesn't have a joystick USB port as that's new with 2.0. Oops. I'm a bit hesitant to remove the 2.0 from a system where it's working perfectly otherwise as a MPU-401 wavetable card and GUS, but I suppose my Pocket386 will now get a PicoGUS (and WiFi!).

crtified-retro, 2025-03-25:
You can enable the joystick USB port functionality on a PicoGUS 1.2, by closing the USBPWR jumper at the top right of the PCB, and using a USB OTG cable (MicroUSB male to USB A female) in the Raspberry Pi Pico's MicroUSB port.

I've never used that function so I can't speak as to it's operational details.

ValadAmoleo, 2025-04-25:
I've just bought a PicoGUS 2.0 and I'm also having joystick woes. With /joy 1 used I still can't get a wired official Xbox 360 controller to work. It should just be a matter of plugging in, doing pgusinit /joy 1 and starting a game (eg Wolfenstein 3D), right?