polpo/picogus #76: DOOM 1/2 1.9 hang on start with GUS mode - VIA MVP3 K6-2+

Lostdotfish, 2025-04-04:
Doom and Doom 2 hang on start when PicoGUS is in GUS mode.

System

EPoX EP-MVP3G 1MB Cache Super Socket 7 (VIA MVP3 chipset)
AMD K6-2+ @600MHz (6x100) - On CPU 128KB cache
256MB SDRAM
DOS 6.22

PicoGUS v2.0
2.2.0 firmware
ULTRASND=240,1,1,5,5

Card is detected and plays audio correctly in Warcraft 2

Doom loads to
`  calling DMX_Init`
then freezes the system.

Let me know if I can help provide any further tests/debug info

Lostdotfish, 2025-04-06:
Additional test today.

Grabbed my 430TX motherboard and moved the PicoGUS over to it.  Booted the exact same DOS 6.22 installation (compact flash to ide).

Doom starts properly and the PicoGUS is working as expected.

So the above issue relates to the Super Socket 7 platform, whether that's the K6-2+ CPU or the VIA MVP3 chipset.

Is there anything I can do to help troubleshoot this?

NeoG2, 2025-06-23:
This appears to be a known issue - Refer to the compatibility chart here:
https://github.com/polpo/picogus/wiki/Compatibility-list

It lists Doom in GUS emulation mode on the MVP3 chipset as replicating a fault that exists on a real GUS

Lostdotfish, 2025-07-03:
> This appears to be a known issue - Refer to the compatibility chart here: https://github.com/polpo/picogus/wiki/Compatibility-list
> 
> It lists Doom in GUS emulation mode on the MVP3 chipset as replicating a fault that exists on a real GUS

Read above.  The same system works fine with real GUS.