polpo/picogus #80: Feature Request - PC Speaker Output

aaaaafuck, 2025-06-28:
Certain older PC games only output to the PC speaker, i.e. Maniac Mansion. While on some other sound cards you can connect the PC speaker to a pin designated for the PC speaker, on the PicoGus you cannot to my knowledge. 

In my setup, I'm running a SoundBlaster Live! and in my single ISA slot I have a PicoGus. Some of these previously mentioned games also have speed issues on higher MHz CPUs. The speed issues can often be fixed with programs like Mo'Slo Deluxe, but these programs only work in truedos mode. The SoundBlaster Live! emulates a SB16 and allows for the PC speaker to be connected, but if you are in truedos it doesn't output any audio because of how it works. The PicoGus on the other hand is very useful for getting sound in games that require truedos mode. Especially with the new CD emulation feature. The problem is that if your only sound card is a PicoGus, there is no way to have the PC Speaker be sent over the line-out. This would require having a second ISA card installed. For this use case, I really think adding a way to output the PC speaker sound through the PicoGus would be helpful to a lot of setups.

Would love to hear if this is possible. Thank you for taking the time to read this.

jharrison03, 2026-03-08:
The Media Vision Pro Audio Spectrum cards were able to sniff the ISA bus and output PC speaker through the card's line-out.  This was done without any cable.  That said, I never managed to actually make this feature work - but the card can apparently do it.

jharrison03, 2026-03-08:
The thing is, I'm building a new machine and if you want to use an older motherboard with a new case, neither the motherboard or the case have a PC speaker anymore.  So it has to be done though the sound card I guess, but that's a somewhat uncommon feature.  I have a CT3670 with a PC speaker input in another machine and I like that feature.