polpo/picogus #81: CD ODE issues when using bat files, long time to start certain games and more

vermian, 2025-07-01:
PC Setup:
Asus P2-99 (Intel 440ZX)
Intel Pentium 2 233MHz @ 166MHz (Klamath) (Slot 1) - 66MHz FSB x 2.5 CPU Multiplier
Voodoo3 3000 (AGP)
PicoGUS (ISA)
Yamaha Audician 32+ (ISA)
128MB SDRAM (1 DIMM)
120GB Intel SSD (SATA) + SATA->IDE Adapter

Using MS-DOS v7.1

Issue 1:
Game: Bedlam (1996)

I have a bat file that mounts "bedlam.cue", then launches the game, however, even when adding a pause in between and waiting 60 seconds, it seems like the game doesn't recognize that the CD is inserted (only thing affected is the lack of CDDA music). If I mount the same image separately, and then launch the game, either the exe directly or through a bat file, the CD seems to be recognized and the CDDA music plays.

bat file looks like this (pgusinit folder is included in path):
```
pgusinit /cdloadname bedlam.cue
pause
bedlam.exe
rem below is for fixing graphics mode on exit
mode 80
```

Issue 2:
I'm sometimes getting a weird issue when I mount a different CD image but for some reason it either doesn't read it properly - folders inaccessible or most files not listed, or sometimes it says it mounted the new image, but the files and music are all from the previously mounted CD image. Seems like restarting the PC fixes this, but not sure how to duplicate yet. 

Issue 3:
Some games take a while to load or even exit.
Rayman (1995) for example takes quite a while to start up (~1 minute) (max install size)
Time Commando (1996) can take a while to start up as well as exit the game (~1 minute each)

I have experienced these issues with 2 different USB sticks so I don't think that's the issue, and if restarting the PC fixes it, I also don't suspect data corruption.

vermian, 2025-07-02:
More DOS Testing with SWIV 3D (1996), it seems like after mounting the bin/cue, the first time you run swiv_dos.exe, it doesn't find the CD, but the second time you run it, it finds it and works fine. Doesn't seem like a timing issue as waiting before launching the exe doesn't seem to make a difference. Oddly, I can use this game as a weird workaround to launch Bedlam through a batch file by having the bat mount the Bedlam bin/cue, change directory to SWIV 3D, launch the game which fails and exits to DOS after a second, then change directory back to Bedlam and load bedlam.exe. Seems like once it tries accessing something on the disc the first time, something changes and the second attempt works? The weird workaround works every time.

vermian, 2025-07-04:
The issue doesn't happen with all games. Yeah I'm using the most recent pgusinit release.

RetroLoom, 2025-07-05:
I tried testing with Bedlam (1996), but so far I am not having any issues. After several tries with loading a different image in between it has loaded every time without issue. 

I used this in my bat file

pgusinit /cdloadname bedlam.cue
f:
bedlam.exe

f: drive being the PicoGUS ODE. Probably unlikely, but it might be good to try refreshing the drive load before loading the file. Otherwise I am not sure how else to test at the moment. 

I am on a Pentium MMX 233Mhz, 128MB, Orpheus 1, so not too far off from the PC specs you are using.

RetroLoom, 2025-07-06:
> It works fine if you run it like that since it runs the bedlam.exe off the CD. You need to run setup.exe and install the files to HD and make the bat run the bedlam.exe from your C: drive. While running it off the CD works, running most of it off the HDD/SSD is much faster for loading.

Ok. I will test from loading it from the HDD and see if that changes anything for me.

RetroLoom, 2025-07-08:
> Oddly enough, even if the same bedlam image is already mounted, if I run the bat which remounts the image, the CDDA doesn't work. I can try a 3rd USB stick but I'm not convinced that is the case.
> 
> Edit: the main issue is the CDDA audio not loading. The game itself still loads fine for me regardless. You are starting the first mission to check, right? The main menu music is not CDDA AFAIK.

Thanks for clarifying this, I understand what you are saying now and can replicate the same issue. CDDA audio is not loading correctly, and sometimes I seem to have other weird issues like the previous loaded image stays loaded instead of bedlam. I will look through some code, but it might be deep to find exactly where this bug would source from. Kevin might have a better idea.

schlockorama, 2025-09-21:
I am using BAT files to swap discs, so I can go easily switch between games. 

The BAT file will say:

PGUSINIT /CDLOADNAME "AITD.CUE"
INDARK

Picogus accepts the command and I see:

CD Image loaded: AITD.CUE

The game will load and crash saying "ERROR: CD Not Found"

I will then go to my CD rom drive which is (O:) and use DIR/P to see what's on the CD drive.

The CD rom drive will show the volume label, BARIS, since I was last playing Buzz Aldrin's Race Into Space before I tried to play Alone in the Dark.

Sometimes the CD rom directory will also be stuck in a directory other than it's root drive.

Let's say I try to play Alone in the Dark again after playing a few games and I see picogus has loaded the image correctly.

Alone in the Dark will now crash in a different area when you select your character and the first voice file is loaded (error message missing voice file).

I will check the CD rom directory again and I'll see O:\AL (since in this case I was last playing Alien Legacy and the drive is stuck in some previous directory that game was using).

I have tried adding "PGUSINIT /CDLOAD 0" before "PGUSINIT /CDLOADNAME "AITD.CUE" to try to unload the drive first and then load a new image, but doesn't seem to help any.

Even if I encounter the error, some games will swap normally like for instance Alien Trilogy doesn't seem to care when using BAT files.

On a fresh computer start, Alone in the Dark 1 will start first time every time with the bat file.

vermian, 2025-09-21:
> I am using BAT files to swap discs, so I can go easily switch between games.
>
> The BAT file will say:
>
> PGUSINIT /CDLOADNAME "AITD.CUE" INDARK
>
> Picogus accepts the command and I see:
>
> CD Image loaded: [...]

I ran into this issue a few times before but I've mostly avoided it for a while. My batch file also switches to USB mode (to make sure the PicoGUS isn't in some other mode) then waits 2 seconds using a program called sleep.exe before loading a CD image (otherwise the PicoGUS isn't ready), so I wonder if setting the mode again on the PicoGUS helps this issue. Also I'm using DOS 7.1. Windows might behave differently since it's a different driver.

schlockorama, 2025-09-22:
> I ran into this issue a few times before but I've mostly avoided it for a while. My batch file also switches to USB mode (to make sure the PicoGUS isn't in some other mode) then waits 2 seconds using a program called sleep.exe before loading a CD image (otherwise the PicoGUS or something else in the chain isn't ready), so I wonder if setting the mode again on the PicoGUS helps this issue. Also I'm using MS-DOS 7.1. Windows might behave differently since it's a different driver.

This is my PC Setup:

120MHZ Pentium 1 (Overclocked to 133MHZ)
Intel Advanced Zappa Socket 5 Motherboard
64 MB RAM
Voodoo 1 3D Graphics Card
S3 Trio 64+ 2D Graphics Card
Soundblaster 16 Multi CD
3 X 8GB CF to IDE Hard Drives (24 GB Total in 12 Partitions)
Picogus 2.0 (Firmware 3.2)

I'm using MS-DOS 6.22. Where can I get that sleep.exe you mentioned? I am using a Launchbox for DOS frontend setup, so it's very annoying to not be able to swap games on the fly consistently. It's the only reason I bought the Picogus was for the CDROM emulation since being able to store all those large bin/cue files and isos off hard drive is very handy when you have a 8GB hard drive motherboard limit.

schlockorama, 2025-09-22:
> My batch file also switches to USB mode (to make sure the PicoGUS isn't in some other mode) then waits 2 seconds using a program called sleep.exe before loading a CD image (otherwise the PicoGUS or [...]

I can consistently get this error to happen on Alone in the Dark 1 for whatever reason (It uses redbook audio maybe?).
The problem will occur without using bat files also.
I tried using the pgusinit /mode usb at the top of the bat file to ensure it's in usb mode and seems to have no effect.
I tried using cdload 0 with no effect.
I tried using cdload n (n is the number for iso or cue) instead of cdloadname and still same issue.
I've used the pause command and waited a long time between swapping discs with no effect so I don't think it's a timing or "cd loading" issue.
There is definately something wrong with the picogus not wanting to change discs.
I've even had it occur right from the start where I try to load the first game.
I might have to go back to the old way of using SCSUCD (can't stream redbook audio, iso only, and takes up hard drive space, but at least consistently works!) to load images for these particular tricky games until a fix is available.