polpo/picogus #85: Feature Request: CD image list with page breaks/pauses

vermian, 2025-07-04:
If you go over a certain amount of CD images and want to list them with "pgusinit /cdlist", it only shows the bottom of the list. Could you add an option to prompt for a key press to go through the CD image list a page at a time?

vermian, 2025-07-07:
Sure, I can try testing it. Thanks!

RetroLoom, 2025-07-07:
You can find my fork and current working branch release [https://github.com/RetroLoom/picogus/releases/tag/v3.1.01beta]here (url)

vermian, 2025-07-08:
I did some testing on your build, though not all features since I already have a SB-compatible card so I can't really use the PicoGUS in that mode, but it seems to work fairly well overall. CD list page breaks work, CDDA volume works. It also fixed a GUS regression in firmware v2/3 that broke the music in Jazz Jackrabbit and Epic Pinball for me.

One little thing that is missing is when you run pgusinit, in v3.1.0 at least, it would state the currently mounted CD image but your build appears to not do that.
I also noticed it defaulted to a main volume of 0 after flashing your build's firmware, is that a bug or on purpose?

vermian, 2025-07-08:
Yeah, it seemed to save properly afterwards.

vermian, 2025-07-13:
Yes, pretty sure, but I can try testing it a bit more. I just noticed it last night.

vermian, 2025-07-15:
> New beta releases with a couple of fixes. [v3.1.2beta](https://github.com/RetroLoom/picogus/releases/tag/v3.1.2beta)

Looks like both new fixes work (only tested CDDA volume though).
I'm hoping you'll also get to making it list CD images in case you type some invalid image name, as discussed earlier.
I was also wondering if there is a CD image limit since it seems like typing anything above "pgusinit /cdload 255" doesn't seem to register as a valid command. With 1-2TB USB drives nowadays I think somebody could reach it, lol.

RetroLoom, 2025-07-17:
> Looks like both new fixes work (only tested CDDA volume though). I'm hoping you'll also get to making it list CD images in case you type some invalid image name, as discussed earlier. I was also [...]

I am thinking about how best to integrate the /cdlist command when the invalid load name is used. I feel like that could become a bit cumbersome for regular use. I have some ideas instead to have it search through the list and try to match the top 3-5 images in the list based on what was entered instead. I wont be able to do this until I have a bit more time though.

polpo, 2025-08-10:
Thanks to @RetroLoom the CD list paging is implemented in firmware v3.2.0: https://github.com/polpo/picogus/releases/tag/v3.2.0.
Regarding the issue with the CD list failing after a failed image load by name, sorry I didn't notice it in this thread, otherwise I would have also addressed that in the v3.2.0 release. Good idea to create an issue for things like that so I have a better chance of noticing them!
About fuzzy matching of loading images, I'm thinking that'd be a good application for a separate interactive CD image management program so pgusinit can remain relatively small and simple.

closed 2025-08-10 by polpo