polpo/picogus #88: Where's the hw/gerbers?

caiot5, 2025-07-18:
The guide for building your own PicoGUS instructs to get the gerbers at hw/gerbers , but those files were deleted. Why?
Where can I find them?

polpo, 2025-07-22:
When I released the v1.2.0 hardware, I moved to putting hardware release artifacts in via the release system. Here is the v1.2.0 release with the gerbers: https://github.com/polpo/picogus/releases/tag/hw_v1.2

BTW, CD-ROM emulation works on all released hardware back to the first publicly released version of the hardware, v1.1.

I'll leave this issue open until I fix the documentation in the wiki on where to find the Gerbers.

caiot5, 2025-07-25:
> When I released the v1.2.0 hardware, I moved to putting hardware release artifacts in via the release system. Here is the v1.2.0 release with the gerbers: [...]

Gotcha, but how would I use the CD-ROM emulation stuff with this v1.2 or v1.1 hardware since there is no USB port?

retrorepair, 2025-07-28:
> When I released the v1.2.0 hardware, I moved to putting hardware release artifacts in via the release system. Here is the v1.2.0 release with the gerbers: [...]

Awesome thank you. I have the DAC and Pico, was considering hacking an old modem to wire it all up but this is much more sensible :)