polpo/picogus #96: unable to change modes with pgusinit Firmware version 3.2.0 card locks up

Joey-boyy, 2025-08-13:
I'm trying the new firmware 3.2.0, and it works great, but when I change modes, I get error: card is not alive after rebooting to new firmware, which locks up the pico, and can only be resolved by powering down the card/system . I am able to change modes, with repeated attempts. I tried running my ISA bus speeds at 5mhz, 6, and 8, with the same results, on 2 different motherboards (a 486 board, and a socket 7 Pentium board), thanks.

polpo, 2025-08-13:
Clocks are configured differently on this firmware so it can be a closer divider of 44100Hz, but it seems that affects its ability to reboot after a mode change. I’ll see if I can find a workaround. Sorry this is happening!

polpo, 2025-08-30:
I have tried to reproduce this issue on 12 different PicoGUS boards, and while 11 don't have any problem, one had a somewhat similar issue – it wouldn't come back up after flashing the new firmware. Only difference is that it also won't start on powerup, but that may be enough for me to figure out a potential workaround for this issue.