polpo/picogus #98: Cannot use /defaults flag PicoGUSinit 3.7.0

saegiru, 2025-08-15:
When running pgusinit /defaults - I get an error "Invalid command /defaults. Run pgusinit /? for usage help"

vermian, 2025-08-15:
I think you have to specify the setting you want to default. From the release notes:
Values can be set back to their default values with a default argument. Example: pgusinit /cdport default
Looks like it's singular "default", not plural.

N4villu5, 2025-09-14:
I'm encountering the same issue.
`pgusinit /?` shows /defaults as one of the options
`/defaults - set all settings for all modes to defaults`
However, running `pgusinit /defaults` will give the output:
`Invalid command /defaults. Run pgusinit /? for usage help`

@vermian I've tried that as well. Running `pgusinit /xyz default` will either do nothing or output the help text depending on what option you put in. Your example `pgusinit /cdport default` gave me the help text.